trvrnrth/esphome-bsec-bme680 #1: Error compiling from hassio

miroslavpetrov, 2020-09-17:
Hello, I am having trouble compiling the esphome-bsec-bme680. I get the following errors:

`INFO Reading configuration /config/esphome/bme680.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/bme680_climate
Processing bme680_climate (board: huzzah; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 15b20e3 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <BSEC Software Library> 1.5.1474 #996a035
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <EEPROM> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/bme680_climate/.pioenvs/bme680_climate/src/esphome/components/api/api_connection.cpp.o
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
*** [/data/bme680_climate/.pioenvs/bme680_climate/src/esphome/components/api/api_connection.cpp.o] Error 1
Compiling /data/bme680_climate/.pioenvs/bme680_climate/src/esphome/components/api/api_pb2.cpp.o
xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'
*** [/data/bme680_climate/.pioenvs/bme680_climate/src/esphome/components/api/api_pb2.cpp.o] Error 1
========================== [FAILED] Took 9.03 seconds ==========================`

trvrnrth, 2020-09-17:
Have you updated to ESPHome 1.15?

This was an issue in the last with 1.14 which meant I had to stick to an older framework version but since updating to 1.15 myself I bumped the framework version and haven't had any issues.

miroslavpetrov, 2020-09-17:
The version I am using is 1.15.1.

trvrnrth, 2020-09-17:
Oddly, now that I look more closely at my compilation logs I think my builds are picking up a cached copy of the framework so I may have inadvertently re-introduced the error here. I'll report back when I've worked out where that is coming from...

trvrnrth, 2020-09-17:
Yep, my bad. I've sorted out the mess I made of updating the patched Arduino framework. It's should now be the correct version!

Unfortunately I don't think this will get pulled automatically so you'll need to force that to happen. The easiest way I'm currently aware of is to shell into your hassio instance, hop onto the esphome docker instance and delete the packages forcing platformio to re-fetch them on the next build:
```
ssh root@homeassistant.local -p22222
login
docker exec -it $(docker ps -f name=esphome -q) bash
rm -Rf ~/.platformio/packages/framework-arduinoespressif8266*
```

This is currently a pretty unpleasant solution anyway. There's a bit of discussion about making all of this nicer in https://github.com/esphome/feature-requests/issues/16#issuecomment-692767632

trvrnrth, 2020-09-18:
OK, so good news. There is now no need for a patched Arduino framework as I've implemented the changes noted for the bsec library packaging in a fork of that repo (in PR to upstream for comment at https://github.com/BoschSensortec/BSEC-Arduino-library/pull/51).

You'll probably still need to clear out the cached framework package files as noted in the last comment but after that a clean build and compile should work as expected without any extra shenanigans.

miroslavpetrov, 2020-09-18:
Thank you! Now it works perfectly! I had a bit of a hard time to find where the I2C address of the BME680 sensor is defined since mine is with default address of 0x77.

closed 2020-09-18 by miroslavpetrov

trvrnrth, 2020-09-18:
Excellent news. Glad it's working for you!

Just in case anyone else is reading this the line to change at the moment is https://github.com/trvrnrth/esphome-bsec-bme680/blob/master/src/bsecsensor.h#L116 with the constant for `0x77` being `BME680_I2C_ADDR_SECONDARY`.

I should probably have made that configurable from the node config so I'll make a note of it for a possible future improvement.

maurik69, 2020-09-23:
> Excellent news. Glad it's working for you!
>
> Just in case anyone else is reading this the line to change at the moment is [...]

That's was really helpful for me too, I'm testing it on ESP32 and I was not able to understand why the lambda was not returning any data nor errors... Thanks!