trvrnrth/esphome-bsec-bme680 #4: How do I adjust the polling interval?

lichtteil, 2020-09-20:
I could not figure out how to change the update interval of the sensor. I think it is now at 15 seconds but I don't understand why :)

lichtteil, 2020-09-21:
Thanks for your reply. Does that mean that the loop in the custom sensor class ( https://github.com/trvrnrth/esphome-bsec-bme680/blob/master/src/bsecsensor.h#L146 ) runs without pause?
I don't need quicker updates but prefer slower updates and have in mind that the lifespan of the sensor expands (at least thats what I think 😄).

trvrnrth, 2020-09-21:
Yes, the loop runs continually with the BSEC library taking care of keeping track of when to actually next poll the sensor in its run function (see https://github.com/trvrnrth/esphome-bsec-bme680/blob/9c5999f60221be149d51697467fc701068cceb49/src/bsecsensor.h#L147).

You could provide a different config at https://github.com/trvrnrth/esphome-bsec-bme680/blob/9c5999f60221be149d51697467fc701068cceb49/src/bsecsensor.h#L124 (see https://github.com/trvrnrth/esphome-bsec-bme680/blob/9c5999f60221be149d51697467fc701068cceb49/src/bsecsensor.h#L6 and https://github.com/BoschSensortec/BSEC-Arduino-library/tree/c5503e0e85067e07e11b45b4215663f929e640c3/src/config) and increase the interval to 5 minutes instead of 3 seconds though I couldn't say what impact a longer interval between sensor heat cycles would have on overall lifetime. Typically you'd do so to reduce power requirements / increase battery lifetime.

lichtteil, 2020-09-22:
Thank you a lot! :)

closed 2020-09-22 by lichtteil