trvrnrth/esphome-bsec-bme680 #6: esp32 reboots sometimes

wettermann32, 2020-11-08:
thanks for the code, things a real easy with it. Anyhow between 30 and 60min runtime the esp32 reboots. IAQ then start from 25 again. I never get a stable value. I have deleted everything from yaml beside bme_680 and wifi/web. Are there any known issues with stability?

Best

trvrnrth, 2020-11-08:
I've certainly had no stability issues running on a couple of ESP8266 Wemos D1 minis. I don't have any ESP32 boards to test with but am not aware of anything that is likely to cause issues there.

Initial calibration can take a fair bit of time, and state is only saved once calibration is achieved so the behavior you describe is expected if you're getting restarts that soon after first boot.

First thing I'd probably try would be to switch on debug logs and see if there's anything odd around the time of reboot then work from there.

wettermann32, 2020-11-08:
First guess, api was forcing a reboot, since no client was connected. Lets see if it survives the night. Thanks for your quick answer anyhow.

closed 2020-11-22 by trvrnrth