trvrnrth/esphome-bsec-bme680 #7: How to setup the update interval

florafrisia, 2020-11-19:
Hello,

I like to thank you the work you did on the bme680.
Still I have a small question I like to setup the update_interval.
Most sensors in ESPHome have this option.

With kind regards,

Peter Noé

trvrnrth, 2020-11-19:
Whilst it would technically be possible to allow a custom update interval it is recommended to run BSEC with one of the standard sampling rates (either every 3 or 300 seconds) see the [explanation from Bosch](https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME680-BSEC-Samping-Rate/m-p/11415/highlight/true#M2333). At present the component does not allow selection of sampling rate and operates with the Low Power 3s mode.

As noted in the readme you may of course apply any further filtering you wish to the sensors and effectively control the update rate in that manner. For example, I apply a [median filter](https://esphome.io/components/sensor/index.html#median) in my setup to guard against outliers and reduce the number of updates sent to home assistant and reduce storage requirements in influxdb which I use to store historical data.

ignisf, 2020-11-20:
Hi @trvrnrth, I'd like to test how the ULP mode behaves, what should I alter in your component's source?

trvrnrth, 2020-11-20:
@ignisf I think that should be it :)

ignisf, 2020-11-21:
@trvrnrth it seems to be working as expected. I also removed the staggering of the sensor state updates. And by 'as expected' I mean that it lowered the load on my esp32cam to a manageable level. I'm still monitoring the measurements.

trvrnrth, 2020-11-21:
@ignisf Good to hear. I may build it in as an option in the future if time allows.

The staggering of the updates was for esp8266 boards really as they couldn't cope with doing all the pushes in one loop call but it shouldn't have any negative impact for other boards with more processing power either. I'm curious why you needed to remove it?

ignisf, 2020-11-21:
I wasn't sure when would the `this->check_bsec_status_() && this->bsec_.run()` condition be true so I wanted to simplify the function as much as possible. Now that I've observed the code work as expected, I may as well add it back.

trvrnrth, 2020-11-21:
@ignisf I'd recommend it. I observed that the pushes blocked which was causing me API disconnects when running on an ESP8266. There's a fair chance you'd not notice on an ESP32 but as it sounds like you've got it doing camera duty as well it'd probably be worth keeping them split just in case!

ariedeleen, 2020-12-10:
As a 'workaound' I have as mentioned earlier set a filter throttle to delay or limit the status updates.
So, now it updates, for example, every five minutes.
Have not figured it out for the text_sensor:
That is still updating every ~3 sec.

[https://esphome.io/components/sensor/index.html?highlight=filters#sensor-filters](url)

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
      filters:
        - throttle: 300s

@trvrnrth thanx for the effort ;-) to get this working with the BSEC shebang.

trvrnrth, 2020-12-11:
@ariedeleen I hadn't really considered the text sensor but you're correct; filters won't work for it. I'll see about getting it to only publish a new state when the value changes.

For the numeric sensors it was a deliberate choice to not have a configurable interval. The underlying sensor run has to be completed regardless and the filters provide much more flexibility.

commit f7b5c150 (2020-12-11): Only publish new text sensor states if changed

This cuts down on unnecessary chatter from the IAQ accuracy text sensor.

Fixes #7
closed 2020-12-11 by trvrnrth