trvrnrth/esphome-bsec-bme680 #8: Code do not compile

Igor-kr, 2020-12-02:
Hello!

Component does not compile.
Throws errors.

```
BSEC Software Library @ 1.6.1480 has been successfully installed!
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <BSEC Software Library> 1.6.1480
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling .pioenvs/blabla/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/api/util.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/bh1750/bh1750.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/blabla/src/esphome/components/bme680_bsec/._bme680_bsec.cpp.o
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/platformio/proc.py", line 46, in run
    self.do_reading()
  File "/home/pi/.local/lib/python3.7/site-packages/platformio/proc.py", line 69, in do_reading
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "/home/pi/.local/lib/python3.7/site-packages/platformio/proc.py", line 69, in <lambda>
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "/usr/lib/python3.7/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd6 in position 96: invalid continuation byte

```
And stops there
```
Compiling .pioenvs/blabla/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/blabla/libFrameworkArduino.a
Indexing .pioenvs/blabla/libFrameworkArduino.a
```

With best regards
Igor

trvrnrth, 2020-12-02:
It looks like you've got a MacOS "resource fork" file in the component directory. Delete `._bme680_bsec.cpp` (and any other `._*` files) and you should be set.

Igor-kr, 2020-12-03:
Hello! 
Ok, now working!

closed 2020-12-03 by Igor-kr