trvrnrth/esphome-bsec-bme680 #13: Enhancement: iaq accuracy as numeric sensor value

martin-at-schorsch, 2020-12-26:
Nice work, thx!

When storing values in a database like influxdb, iaq accuracy as numeric sensor value in addition to text would be nice.

wifwucite, 2020-12-29:
You should be able to add that via configuration: take a look at the template sensor. You can specify a lambda that takes the current state of the iaq accuracy text sensor and converts it to a numeric value.

trvrnrth, 2020-12-31:
As @wifwucite suggests a lambda would work nicely, but since the changes to support an optional numeric sensor are pretty small I'll add that option too as it's a little bit neater :)

itsyoshio, 2021-01-02:
Yup, i'd like a numeric sensor aswell.

Would a 0-100% scale make sense in terms of Air Quality Index or would it "just" be an int ranging from 0 to 500?
Maybe there's another scale that would be more meaningful, but my guess is that an Air Quality of >= 50% should alarm any Person.

trvrnrth, 2021-01-02:
@itsyoshio The IAQ is available as a numeric sensor already which outputs in the range 0 to 500. This issue is about exposing the **accuracy** as a numeric sensor in addition to the existing text sensor. Your graph setup looks like it works well.

closed 2021-01-03 by trvrnrth

mgschwandtner, 2021-01-04:
Thx for adding the feature!