trvrnrth/esphome-bsec-bme680 #14: Enhancement: Make sample configurable

wifwucite, 2020-12-29:
In BME680BSECComponent::setup the sample rate is fixed to BSEC_SAMPLE_RATE_LP. It would be nice to have an optional configuration which could change it to BSEC_SAMPLE_RATE_ULP to save power.

gretel, 2020-12-31:
maybe this is worth mentioning https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BME680-BSEC-Samping-Rate/td-p/11413

trvrnrth, 2020-12-31:
@gretel Yep; I've started adding support for the ULP mode but don't intend to add support for custom intervals on the basis of that guidance.

gretel, 2020-12-31:
@trvrnrth thanks. i wasn't intending to say it's required or even useful :)

closed 2021-01-04 by trvrnrth

wifwucite, 2021-01-04:
Thanks for the quick enhancement.

trvrnrth, 2021-01-04:
@wifwucite No problem; this one was fairly straighforward. Hope it works out for you :)

gretel, 2021-01-04:
[image]
@trvrnrth works for me. thank you!

trvrnrth, 2021-01-05:
@gretel Liking your choice of IAQ icon!