trvrnrth/esphome-bsec-bme680 #16: Failed configuration of bme680

thobastian, 2020-12-31:
Hi !
I copied the files to my ESPhome-dir and copied your code into my config-file. 
Can you help ?
Thanks !!

I got these error-messages:

`INFO Reading configuration /config/esphome/katzenklo.yaml...
Failed config

bme680_bsec: [source /config/esphome/katzenklo.yaml:32]
  
  Component not found: bme680_bsec.
  address: 119
  temperature_offset: 0.9
  iaq_mode: mobile
  state_save_interval: 4h
sensor.bme680_bsec: [source /config/esphome/katzenklo.yaml:44]
  
  Platform not found: 'sensor.bme680_bsec'.
  platform: bme680_bsec
  temperature: 
    name: BME680 Temperature
  pressure: 
    name: BME680 Pressure
  humidity: 
    name: BME680 Humidity
  gas_resistance: 
    name: BME680 Gas Resistance
  iaq: 
    name: BME680 IAQ
  co2_equivalent: 
    name: BME680 CO2 Equivalent
  breath_voc_equivalent: 
    name: BME680 Breath VOC Equivalent
text_sensor.bme680_bsec: [source /config/esphome/katzenklo.yaml:61]
  
  Platform not found: 'text_sensor.bme680_bsec'.
  platform: bme680_bsec
  iaq_accuracy: 
    name: BME680 IAQ Accuracy
`

caddyblack, 2020-12-31:
Hi,

Try address 0x76.

thobastian, 2020-12-31:
sorry, but that don't works. 
I try this example and that works:
sensor:
  - platform: bme680
    temperature:
      name: "BME680 Temperature"
      oversampling: 16x
    pressure:
      name: "BME680 Pressure"
    humidity:
      name: "BME680 Humidity"
    gas_resistance:
      name: "BME680 Gas Resistance"
    address: 0x77
    update_interval: 60s

thobastian, 2020-12-31:
perhaps I use the wrong directory of "bme680_bsec" ?

caddyblack, 2020-12-31:
So your working address is 0x77, not the default 0x76.
In your example you use address 116.

thobastian, 2020-12-31:
yes, I use 119. but hexadecimal is this 0x77.

thobastian, 2020-12-31:
now, the directory is correct! :-)
but I got more errors:

`ERROR Unable to load custom component bme680_bsec:
Traceback (most recent call last):
  File "/opt/esphome/esphome/config.py", line 134, in _lookup_module
    module = importlib.import_module(f'custom_components.{domain}')
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 781, in get_code
  File "<frozen importlib._bootstrap_external>", line 741, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/config/esphome/custom_components/bme680_bsec/__init__.py", line 7
    <!DOCTYPE html>
    ^
SyntaxError: invalid syntax
ERROR Unable to load custom component bme680_bsec.sensor:
Traceback (most recent call last):
  File "/opt/esphome/esphome/config.py", line 134, in _lookup_module
    module = importlib.import_module(f'custom_components.{domain}')
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 941, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 781, in get_code
  File "<frozen importlib._bootstrap_external>", line 741, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/config/esphome/custom_components/bme680_bsec/__init__.py", line 7
    <!DOCTYPE html>
    ^
SyntaxError: invalid syntax
ERROR Unable to load custom component bme680_bsec.text_sensor:
Traceback (most recent call last):
  File "/opt/esphome/esphome/config.py", line 134, in _lookup_module
    module = importlib.import_module(f'custom_components.{domain}')
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 941, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 781, in get_code
  File "<frozen importlib._bootstrap_external>", line 741, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/config/esphome/custom_components/bme680_bsec/__init__.py", line 7
    <!DOCTYPE html>
    ^
SyntaxError: invalid syntax`

caddyblack, 2020-12-31:
Hi,

Did you define thebme680_bsec in your script?
```

bme680_bsec:
    # i2c address override (default is 0x76)
    address: 0x77

    # Temperature offset if device is in enclosure and reads too high (default is 0)
    temperature_offset: 1.6

    # Mode for IAQ sensors if device is mobile (default is static)
    iaq_mode: static

    # Interval at which to save BSEC state (default is 6 hours)
    state_save_interval: 6h

Sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
    pressure:
      name: "BME680 Pressure"
    humidity:
      name: "BME680 Humidity"
    gas_resistance:
      name: "BME680 Gas Resistance"
    iaq:
      name: "BME680 IAQ"
    co2_equivalent:
      name: "BME680 CO2 Equivalent"
    breath_voc_equivalent:
      name: "BME680 Breath VOC Equivalent"

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      name: "BME680 IAQ Accuracy"

```

thobastian, 2020-12-31:
No, who can I do this ?

caddyblack, 2020-12-31:
Make it part of your .yaml file.

Example of my .yaml file  ( I use an esp6288 to communicate to the sensor and homeassistant to show the values):
```
esphome:
  name: co2
  platform: ESP8266
  board: nodemcu

wifi:
  ssid: !secret netwerk
  password: !secret netwerkID

  manual_ip:
    static_ip: 192.168.2.XX
    gateway: 192.168.2.254
    subnet: 255.255.255.0

    
 # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "CO2 Fallback Hotspot"
    password: "mypassword"

time:
  - platform: homeassistant
    id: esptime

i2c:
  - id: bme
    sda: D7
    scl: D5
    scan: true

bme680_bsec:
    # i2c address override (default is 0x76)
    address: 0x77

    # Temperature offset if device is in enclosure and reads too high (default is 0)
    temperature_offset: 1.6

    # Mode for IAQ sensors if device is mobile (default is static)
    iaq_mode: static

    # Interval at which to save BSEC state (default is 6 hours)
    state_save_interval: 6h

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
      filters:
       - sliding_window_moving_average:
          window_size: 30
          send_every: 30
    pressure:
      name: "BME680 Pressure"
      filters:
       - sliding_window_moving_average:
          window_size: 30
          send_every: 30
    humidity:
      name: "BME680 Humidity"
      filters:
       - sliding_window_moving_average:
          window_size: 30
          send_every: 30
    gas_resistance:
      name: "BME680 Gas Resistance"
      filters:
       - sliding_window_moving_average:
          window_size: 30
          send_every: 30
    iaq:
      name: "BME680 IAQ"
      filters:
       - sliding_window_moving_average:
          window_size: 30
          send_every: 30
    co2_equivalent:
      name: "BME680 CO2 Equivalent"
      filters:
       - sliding_window_moving_average:
          window_size: 30
          send_every: 30
    breath_voc_equivalent:
      name: "BME680 Breath VOC Equivalent"
      filters:
       - sliding_window_moving_average:
          window_size: 30
          send_every: 30

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      name: "BME680 IAQ Accuracy"

captive_portal:

# Enable logging
logger:
  level: DEBUG

api:
  password: !secret devpw

ota:
  password: !secret devpw

web_server:
  port: 80
  auth:
    username: !secret user
    password: !secret passwd
```

thobastian, 2020-12-31:
Now, it works.  
I copied the files wrong. Now, Idownload the zip-file and it works fine.  
thanks for your help !

and have a nice Sylvester
Thomas

closed 2020-12-31 by thobastian

caddyblack, 2020-12-31:
Hi ,

I am glad you got it to work.
Are you going to sniff the cats loo ( katzenklo) with it?

I find the co2 values off from reality a lot.
After a month the values become better ( also the actual measurement of
resistivity used to calculate CO2 went up from 13kohm first day to 140kohm
after a month).
Looks like the bme680 needs a long burn-in time.
I use the default 300 degC and 150ms. Playing with those numbers did not
hel much.
I use an mz-19 as reference.

Enjoy the sensor and have a great 2021!

trvrnrth, 2020-12-31:
@thobastian Glad you got things up and running :) Sounds like you have an interesting use case!

@caddyblack Thanks for the support. I've updated the readme to make it a bit clearer where the directory should be placed in 251615f298e4659a41df02a5b47ad99a03196d88

fradaxx, 2022-05-11:
I have the same problem with an ESP8266 module  that I try to use with ESPHome in Home Assistant and, if possible, I would like to ask for some help to try to solve it.

My configuration file (livingroom.yaml) is the following:

`esphome:
  name: livingroom

esp8266:
  board: esp12e

external_components:
  - source: github://trvrnrth/esphome-bsec-bme680

# Enable logging
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Livingroom Fallback Hotspot"
    password: "dm@fnc68r15"

captive_portal:
    

# Enable Home Assistant API
api:

ota:
  password: "************"

# Example configuration entry
i2c:
  sda: GPIO4
  scl: GPIO5
  scan: True
  id: bus_a
  
switch:
  - platform: gpio
    id: ceiling_lamp_relay
    pin: GPIO13
    inverted: yes # to correctly control commonly available relay modules with active low inputs
    restore_mode: RESTORE_DEFAULT_OFF
  - platform: template
    name: "ceiling lamp relay"
    icon: "mdi:lightbulb"
    turn_on_action:
     - switch.turn_on: ceiling_lamp_relay
     - delay: 500ms
     - switch.turn_off: ceiling_lamp_relay

binary_sensor:
  - platform: gpio
    name: "ceiling_lamp_status"
    pin: 
      number: GPIO12
      mode: INPUT_PULLUP
      inverted: false

bme680_bsec:
    # i2c address
    # -----------
    # Common values are:
    # - 0x76
    # - 0x77
    # Default: 0x76
[... 136 lines cut ...]
    filters: 
      - median
  breath_voc_equivalent: 
    name: BME680 Breath VOC Equivalent
    filters: 
      - median`

I've also tried to add the 

`external_components:
  - source: github://trvrnrth/esphome-bsec-bme680`

key to the same file but without success; in any case I see a red X on the 

 `  - platform: bme680_bsec`

line and when I do the validation step, I obtain the previous errors.

The validation and update process works well if I remove the BME680 sensor block and the other features of the module are visible and work well in Home Assistant.

ESPHome is updated to the current release (2022.4.0).

Please, is it possible to have some hint on how to solve this problem?

Any help will be appreciated! Thanks

trvrnrth, 2022-05-11:
You don't need to copy files over or include as an external component as it is now [included in the core distribution](https://github.com/trvrnrth/esphome-bsec-bme680#gift-component-now-included-in-the-core-distribution-gift).

fradaxx, 2022-05-11:
> You don't need to copy files over or include as an external component as it is now [included in the core distribution](https://github.com/trvrnrth/esphome-bsec-bme680#gift-component-now-included-in-the-core-distribution-gift).

Thank for your answer; It is not clear to me if the core distribution includes only the support for the basic features of the BME680 or also Bosch's BSEC custom library. In any case, the error at the validate step persists.

Please, can you post a demo of a script using the BSEC library to show the IAQ and CO2 equivalent readings?

trvrnrth, 2022-05-11:
@fradaxx If you follow the link there is [full documentation available](https://esphome.io/components/sensor/bme680_bsec.html).

You will need to remove the custom component directory if you still have one and may need to clean your build files.

fradaxx, 2022-05-11:
I removed the bme680_bsec subfolder in the custom_components folder and, following the online guide, I have modified the script as follows:

`esphome:
  name: livingroom

esp8266:
  board: esp12e

# Enable logging
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Livingroom Fallback Hotspot"
    password: "************"

captive_portal:
    

# Enable Home Assistant API
api:

ota:
  password: "dm@fnc68r15"

switch:
  - platform: gpio
    id: ceiling_lamp_relay
    pin: GPIO13
    inverted: yes # to correctly control commonly available relay modules with active low inputs
    restore_mode: RESTORE_DEFAULT_OFF
  - platform: template
    name: "ceiling lamp relay"
    icon: "mdi:lightbulb"
    turn_on_action:
     - switch.turn_on: ceiling_lamp_relay
     - delay: 500ms
     - switch.turn_off: ceiling_lamp_relay

binary_sensor:
  - platform: gpio
    name: "ceiling_lamp_status"
    pin: 
      number: GPIO12
      mode: INPUT_PULLUP
      inverted: false

# Example configuration entry
i2c:

# Minimal example configuration with common sensors
bme680_bsec:

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
    pressure:
      name: "BME680 Pressure"
    humidity:
      name: "BME680 Humidity"
    iaq:
      name: "BME680 IAQ"
    co2_equivalent:
      name: "BME680 CO2 Equivalent"
    breath_voc_equivalent:
      name: "BME680 Breath VOC Equivalent"

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:`

When I try to clean the build files I obtain the following messages:

`INFO Reading configuration /config/esphome/livingroom.yaml...
Failed config

bme680_bsec: [source /config/esphome/livingroom.yaml:58]
  
  Component bme680_bsec cannot be loaded via YAML (no CONFIG_SCHEMA).
  {}
sensor.bme680_bsec: [source /config/esphome/livingroom.yaml:61]
  
  Platform not found: 'sensor.bme680_bsec'.
  platform: bme680_bsec
  temperature: 
    name: BME680 Temperature
  pressure: 
    name: BME680 Pressure
  humidity: 
    name: BME680 Humidity
  iaq: 
    name: BME680 IAQ
  co2_equivalent: 
    name: BME680 CO2 Equivalent
  breath_voc_equivalent: 
    name: BME680 Breath VOC Equivalent
text_sensor.bme680_bsec: [source /config/esphome/livingroom.yaml:76]
  
  Platform not found: 'text_sensor.bme680_bsec'.
  platform: bme680_bsec
  iaq_accuracy: 
    name: BME680 IAQ Accuracy`

and, unfortunately, the error with the bme680_bsec platform at the update step persists.

trvrnrth, 2022-05-11:
Are you running a recent version of ESPHome?

fradaxx, 2022-05-11:
Yes, it is the 2022.4.0; I freshly reinstalled it yesterday.

trvrnrth, 2022-05-11:
@fradaxx That's strange then. You might want to hit up the [ESPHome discord](https://discord.com/channels/429907082951524364/429907082955718657) to see if anyone has any ideas what might be wrong with your setup. You could also try manually cleaning out the build dir from `.esphome`.

fradaxx, 2022-05-11:
@trvrnrth thank you for your willingness; I'll try both solutions!