trvrnrth/esphome-bsec-bme680 #17: including the bsec lib

FlorianLudwig, 2020-12-31:
Hi,

how do I include the bsec lib into the esphome build process?

I am getting the following error:

```
In file included from src/esphome/components/bme680_bsec/bme680_bsec.cpp:3:0:
src/esphome/components/bme680_bsec/bme680_bsec.h:10:18: fatal error: bsec.h: No such file or directory
```

Thanks :)

FlorianLudwig, 2020-12-31:
Solution:

```
esphome:
  platformio_options:
    lib_deps:
     - "boschsensortec/BSEC-Arduino-library"
 ```

The important part is to not add a `@ version` at the end. Otherwise it fails.

closed 2020-12-31 by FlorianLudwig

trvrnrth, 2020-12-31:
@FlorianLudwig It shouldn't be necessary to manually add any dependencies at all as the component itself specifies the required version via esphome codegen: https://github.com/trvrnrth/esphome-bsec-bme680/blob/f7b5c150e3453cf22047d3a01426f31661e9ce95/bme680_bsec/__init__.py#L45

You should end up with something like the following on a clean build:
```
INFO Reading configuration /config/esphome/test_bme680_bsec.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test_bme680_bsec
Processing test_bme680_bsec (board: d1_mini; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Looking for ESPAsyncTCP-esphome library in registry
LibraryManager: Installing id=6757 @ 1.2.3
ESPAsyncTCP-esphome @ 1.2.3 has been successfully installed!
Looking for ESPAsyncWebServer-esphome library in registry
LibraryManager: Installing id=6758 @ 1.2.7
ESPAsyncWebServer-esphome @ 1.2.7 has been successfully installed!
Looking for ESPAsyncTCP-esphome library in registry
LibraryManager: Installing id=6757
ESPAsyncTCP-esphome @ 1.2.3 has been successfully installed!
Looking for AsyncTCP-esphome library in registry
LibraryManager: Installing id=6798
AsyncTCP-esphome @ 1.1.1 has been successfully installed!
Looking for BSEC Software Library library in registry
LibraryManager: Installing id=6979 @ 1.6.1480
BSEC Software Library @ 1.6.1480 has been successfully installed!
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <BSEC Software Library> 1.6.1480
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
```