trvrnrth/esphome-bsec-bme680 #22: support [id] value for ifclause

itsyoshio, 2021-01-02:
I'd be nice if the Library could support [id] to be able to do things like conditionals on the text_sensor's state.

```yaml
    on_value:
      then:
        - if:
          condition:
            or: 
              - text_sensor.state:
                id: iaq_accuracy
                state: 'Uncertain'
```

ESPHome Docs for this feature are found [here](https://esphome.io/components/text_sensor/index.html)

trvrnrth, 2021-01-02:
You're putting the id in the wrong place. The following ought to work:
```yaml
text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      # IAQ accuracy as a text value of Stabilizing, Uncertain, Calibrating, Calibrated
      id: iaq_accuracy
      name: "BME680 IAQ Accuracy"
    on_value:
      then:
        - if:
          condition:
            or: 
              - text_sensor.state:
                id: iaq_accuracy
                state: 'Uncertain'
```

itsyoshio, 2021-01-04:
Thanks, works like a charm!
Sorry about this.

closed 2021-01-04 by itsyoshio