trvrnrth/esphome-bsec-bme680 #25: i2c address 0x77 - Error and Safe Mode

dmar29, 2021-01-07:
Hello, it´s my first time, that i write an issue on github. So sorry when i m doing it the wrong way. My BME680 comes with address 0x77. When i do so in the YAML-File, I got an Error-Message and the ESP goes in the Safe Mode. When i change the Config to 0x76 i have no Errors, but also no sensor-readings :-)

the normal config from esphome.io works with the address 0x77. so maybe it´s something in the bsec-framework? I just found a "hard entry" 0x76 in the init-File.

Sorry, that i made no Screenshots.

trvrnrth, 2021-01-07:
I'm not sure what "hard entry" you're referring to but if you can provide your config and log output then I may be able to help you.

dmar29, 2021-01-08:
> I'm not sure what "hard entry" you're referring to but if you can provide your config and log output then I may be able to help you.

This part in your init-File:
```
    cv.Optional(CONF_STATE_SAVE_INTERVAL, default='6hours'): cv.positive_time_period_minutes,
}).extend(i2c.i2c_device_schema(0x76))
```

i copy all of your example code in my configuration and this is the log-file:
```
[09:37:12][I][i2c:033]: Scanning i2c bus for active devices...
[09:37:12][I][i2c:040]: Found i2c device at address 0x3C
[09:37:12][I][i2c:040]: Found i2c device at address 0x77
[09:37:12][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PIR Sensor'
[09:37:12][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[09:37:12][C][gpio.binary_sensor:016]:   Pin: GPIO14 (Mode: INPUT)
[09:37:12][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Helligkeit'
[09:37:12][C][gpio.binary_sensor:016]:   Pin: GPIO12 (Mode: INPUT, INVERTED)
[09:37:12][C][logger:185]: Logger:
[09:37:12][C][logger:186]:   Level: DEBUG
[09:37:12][C][logger:187]:   Log Baud Rate: 115200
[09:37:12][C][logger:188]:   Hardware UART: UART0
[09:37:12][C][bme680_bsec.sensor:059]: BME680 via BSEC:
[09:37:12][C][bme680_bsec.sensor:061]:   BSEC Version: 1.4.8.0
[09:37:12][C][bme680_bsec.sensor:062]:   Address: 0x76
[09:37:12][E][bme680_bsec.sensor:066]: Communication failed with BSEC Status: 0, BME680 Status: -2
```
i think this is correct, because my sensor is at 0x77 and not 0x76.
Here is the cofiguration with Adress 0x77:

```
esphome:
  name: senspack_buero
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "123"
  password: "123"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Senspack Buero Fallback Hotspot"
    password: "123"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "123"

ota:
  password: "123"

    
# i2c-Bus für die Übertragung ans Display und BME680
i2c:
  sda: D1
  scl: D2
 
bme680_bsec:
    # i2c address
    # -----------
    # Common values are:
    # - 0x76
    # - 0x77
    # Default: 0x76
    address: 0x77

    # Temperature offset
    # ------------------
    # Useful if device is in enclosure and reads too high
    # Default: 0
    temperature_offset: 0

    # IAQ calculation mode
    # --------------------
    # Available options:
    # - static (for fixed position devices)
    # - mobile (for on person or other moveable devices)
    # Default: static
    iaq_mode: static

    # Sample rate
    # -----------
    # Available options:
    # - lp (low power - samples every 3 seconds)
    # - ulp (ultra low power - samples every 5 minutes)
    # Default: lp
    sample_rate: lp

    # Interval at which to save BSEC state
    # ------------------------------------
    # Default: 6h
    state_save_interval: 6h

sensor:
  - platform: bme680_bsec
    temperature:
      # Temperature in °C
      name: "BME680 Temperature"
      filters:
        - median
    pressure:
      # Pressure in hPa
      name: "BME680 Pressure"
      filters:
        - median
    humidity:
      # Relative humidity %
      name: "BME680 Humidity"
      filters:
        - median
    gas_resistance:
      # Gas resistance in Ω
      name: "BME680 Gas Resistance"
      filters:
        - median
    iaq:
      # Indoor air quality value
      name: "BME680 IAQ"
      filters:
        - median
    iaq_accuracy:
      # IAQ accuracy as a numeric value of 0, 1, 2, 3
      name: "BME680 Numeric IAQ Accuracy"
    co2_equivalent:
      # CO2 equivalent estimate in ppm
      name: "BME680 CO2 Equivalent"
      filters:
        - median
    breath_voc_equivalent:
      # Volatile organic compounds equivalent estimate in ppm
      name: "BME680 Breath VOC Equivalent"
      filters:
        - median

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      # IAQ accuracy as a text value of Stabilizing, Uncertain, Calibrating, Calibrated
      name: "BME680 IAQ Accuracy"

# Bewegungsmelder einrichten
binary_sensor:
  - platform: gpio
    pin: D5
    name: "PIR Sensor"
    device_class: motion
    id: motion
    
# Sensor für Helligkeit mit digitaler Ausgabe (1/0)
# Es wird erst nach 100 ms auf Dunkel geschaltet, um ggf. kurze Abdeckungen zu vermeiden.
  - platform: gpio
    pin: 
      number: D6
      inverted: True
    name: "Helligkeit"
    filters:
      - delayed_off: 500ms

    
# Font für Display festlegen
font:
  - file: "Tahoma Regular font.ttf"
    id: my_font
    size: 12

# Display für die Anzeige
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x32"
    reset_pin: D0
    address: 0x3C
    lambda: |-
      if (id(motion).state) {
      it.printf(0, 4, id(my_font), "Hallo und" );
      it.printf(0, 20, id(my_font), "Guten Tag" );
      }

# ALTCODE:
# Anzeige vom DHT22
#      if (id(motion).state) {
#      it.printf(0, 4, id(my_font), "Temp: %.1f°C", id(temperature).state);
#      it.printf(0, 20, id(my_font), "LF: %.1f%%", id(humidity).state);
#      }
```

and the log-file after upload:
```
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from senspack_buero.local using esphome API
INFO Connecting to senspack_buero.local:6053 (192.168.178.24)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.178.24: timed out). Re-Trying in 1 seconds
INFO Connecting to senspack_buero.local:6053 (192.168.178.24)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.178.24: timed out). Re-Trying in 1 seconds
INFO Connecting to senspack_buero.local:6053 (192.168.178.24)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.178.24: [Errno 111] Connection refused). Re-Trying in 2 seconds
```

trvrnrth, 2021-01-08:
The value you see in the init is just used to set the default:
https://github.com/esphome/esphome/blob/5c86f332b269fd3e4bffcbdf3359a021419effdd/esphome/components/i2c/__init__.py#L37

Do you get any more of the decoded stack trace in your USB log or is that the end of the output?

It would also be interesting to know if your setup works if you temporarily disable your other i2c display device.

dmar29, 2021-01-08:
it works without the display.

trvrnrth, 2021-01-08:
Interesting. I don't have an SSD1306 on hand to test with so I've ordered one. I've been meaning to have a play with the display stuff anyway so it's as good a device to start with as any! I'll report back once I have it and have had a chance to do some testing.

caddyblack, 2021-01-08:
Just some information:
I had similar event. After removing the display from the configuration it worked.
I was **assuming** the display consumed too much processing time so the BME680 information was not processed correctly.
I have no proof of it.
I do have an MHZ-19 CO2 sensor in the same configuration as a reference. That combination is OK.
All connected to a NodeMCU ESP8266.

trvrnrth, 2021-01-08:
@caddyblack Thanks; that's useful to know. Your assumption sounds entirely plausible to me. It might be that the SSD1306 component needs to yield during it's write of display data.

SenexCrenshaw, 2021-01-08:
Maybe @kbx81  can help. He helped with the latest code for the SSD1306

trvrnrth, 2021-01-17:
@dmar29 Turns out it's not the display itself or the comms to it that are the issue here; it's loading in all the font glyphs that pushed things over the edge. I've made some improvements in #27 to reduce the RAM requirements of this component which has got things up and running for me on a test ESP8266 board.

You may also need to limit the glyphs loaded (see [config](https://esphome.io/components/display/index.html?highlight=font#drawing-static-text)) depending on what else you have loaded / if you add more fonts / images etc.

trvrnrth, 2021-01-17:
@dmar29 I'd be grateful if you could try that branch out and let me know if that resolves things for you without any other changes :)

trvrnrth, 2021-01-17:
Just for fun I stuck together a quick little example config to show some of the sensor output:
[image]

```sensor:
  - platform: bme680_bsec
    temperature:
      id: temperature
      name: "BME680 Temperature"
    pressure:
      id: pressure
      name: "BME680 Pressure"
    humidity:
      id: humidity
      name: "BME680 Humidity"
    gas_resistance:
      name: "BME680 Gas Resistance"
    iaq:
      id: iaq
      name: "BME680 IAQ"
    iaq_accuracy:
      id: iaq_accuracy_numeric
      name: "BME680 IAQ Accuracy Numeric"
    co2_equivalent:
      id: co2
      name: "BME680 CO2 Equivalent"
    breath_voc_equivalent:
      id: voc
      name: "BME680 Breath VOC Equivalent"

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      id: iaq_accuracy
      name: "BME680 IAQ Accuracy"

font:
  - file: "tahoma.ttf"
    id: tahoma_23
    size: 23
    glyphs: "0123456789."
  - file: "tahoma.ttf"
    id: tahoma_10
    size: 10

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    lambda: |-
      it.printf(0, -7, id(tahoma_23), "%.1f", id(temperature).state);
      it.print(47, -2, id(tahoma_10), "°C");

      it.printf(70, -7, id(tahoma_23), "%.1f", id(humidity).state);
      it.print(117, -2, id(tahoma_10), "%");

      it.printf(0, 21, id(tahoma_10), "IAQ: %.0f (%s)", id(iaq).state, id(iaq_accuracy).state.c_str());
      it.printf(0, 32, id(tahoma_10), "VOC: %.1f ppm", id(voc).state);
      it.printf(0, 43, id(tahoma_10), "CO2: %.1f ppm", id(co2).state);
      it.printf(0, 54, id(tahoma_10), "Pressure: %.1f hPa", id(pressure).state);
```

dmar29, 2021-01-17:
i tested it a view minutes ago. Works fine. Thank you.

closed 2021-01-17 by dmar29

trvrnrth, 2021-01-17:
Awesome!