trvrnrth/esphome-bsec-bme680 #28: Potential bug - no sensor updates

sjtrny, 2021-01-18:
Hi, I've started noticing that when I use this component the sensor seems to get stuck and doesn't send out updates. I switched back to the standard esphome BME680 component and it doesn't happen.

Symptoms:
- home assistant doesn't receive updates from the sensors for many hours
- other components on the esphome are updating normally
- it resumes normal operation after a reboot

[image]

trvrnrth, 2021-01-18:
Nothing in the logs I assume? The [status check function](https://github.com/trvrnrth/esphome-bsec-bme680/blob/f7f46d7a15010895ae80c3ab56dc520542063c20/bme680_bsec/bme680_bsec.cpp#L146) should output errors from the library if any are encountered.

sjtrny, 2021-01-19:
Nothing that I could see. I've switched over to the standard component on the device since it seems more reliable, which is critical since it is in a greenhouse with sensitive plants.

I'll setup a test rig on my bench to see if I can recreate the problem again and catch an error message.

sjtrny, 2021-01-21:
I can't replicate it with a single BME680 on an ESP32. It might be conflicting with the Dallas sensor that is on the board. I'm closing this issue for now. I'm not sure I will have time to do further testing for a while.

closed 2021-01-21 by sjtrny

trvrnrth, 2021-01-21:
I reduced the memory requirements for the component the other day; but it's pretty unlikely that that's the issue here as that only cropped up when loading in font glyphs for a display. As a result of that investigation I've got a test setup running on the desk by me at the moment driving an I2C display at the same time and that seems happy enough but that's fairly anecdotal. I'd perhaps be inclined to just run a single sensor per board but I don't know how silly that might get for your setup!

Anyway; thanks for the update. If you do end up with time for more testing in the future and have any more info please feel free to re-open.