trvrnrth/esphome-bsec-bme680 #30: use of "median" filters as shown in the example lead to 'Components should block for at most 20-30ms in loop().' warning

gretel, 2021-01-24:
one more thing i noted while checking on https://github.com/trvrnrth/esphome-bsec-bme680/issues/29:

```
[21:59:54][V][sensor:013]: 'PoniWormSuite Temperature': Received new state 19.305878
[21:59:54][VV][sensor.filter:013]: Filter(0x3ffb2820)::input(19.305878)
[21:59:54][VV][sensor.filter:053]: MedianFilter(0x3ffb2820)::new_value(19.305878)
[21:59:54][V][sensor:013]: 'PoniWormSuite Humidity': Received new state 44.856663
[21:59:54][VV][sensor.filter:013]: Filter(0x3ffb3178)::input(44.856663)
[21:59:54][VV][sensor.filter:053]: MedianFilter(0x3ffb3178)::new_value(44.856663)
[21:59:54][V][sensor:013]: 'PoniWormSuite Pressure': Received new state 996.859985
[21:59:54][VV][sensor.filter:013]: Filter(0x3ffb2ddc)::input(996.859985)
[21:59:54][VV][sensor.filter:053]: MedianFilter(0x3ffb2ddc)::new_value(996.859985)
[21:59:54][V][sensor:013]: 'PoniWormSuite Gas Resistance': Received new state 104816.000000
[21:59:54][VV][sensor.filter:013]: Filter(0x3ffb356c)::input(104816.000000)
[21:59:54][VV][sensor.filter:053]: MedianFilter(0x3ffb356c)::new_value(104816.000000)
[21:59:54][V][sensor:013]: 'PoniWormSuite IAQ': Received new state 25.000000
[21:59:54][VV][sensor.filter:013]: Filter(0x3ffb3930)::input(25.000000)
[21:59:54][VV][sensor.filter:053]: MedianFilter(0x3ffb3930)::new_value(25.000000)
[21:59:54][V][sensor:013]: 'PoniWormSuite CO2 Equivalent': Received new state 500.000000
[21:59:54][VV][sensor.filter:013]: Filter(0x3ffb3e14)::input(500.000000)
[21:59:54][VV][sensor.filter:053]: MedianFilter(0x3ffb3e14)::new_value(500.000000)
[21:59:54][V][sensor:013]: 'PoniWormSuite VOC Equivalent': Received new state 0.500000
[21:59:54][VV][sensor.filter:013]: Filter(0x3ffb3cd8)::input(0.500000)
[21:59:54][VV][sensor.filter:053]: MedianFilter(0x3ffb3cd8)::new_value(0.500000)
[21:59:54][V][app:081]: A component took a long time in a loop() cycle (0.74 s).
[21:59:54][V][app:082]: Components should block for at most 20-30ms in loop().
````

again, this might be more of an upstream issue. but probably worth raising your attention. regards!

trvrnrth, 2021-01-25:
Looking at the core app code I think that's the loop execution time for *all* components so the message appears to be slightly misleading.

The upstream BSEC library does necessitate use of a delay in the loop which isn't ideal but I believe it will be relatively short and will at least yield allowing things like the TCP stack to be serviced.

I'm also not sure if you're seeing the effects of that or the time taken to push the sensor states themselves. I did previously have the state pushes split across subsequent loop calls but neatened that up in 8226e17eeb97138d08948e78d38e5d0a8de48f7b as it was pretty ugly. I'm not actually sure if the pushes via the API are decoupled further up the chain internally so it's possible that network latency might come into play here too.

0.74 seconds does seem very high though so when I get a chance I'll check the behaviour on my test board and see how that compares.

gretel, 2021-01-25:
@trvrnrth thanks for the insight!

> it's possible that network latency might come into play here too

good point. it's actually far worse than i expected:

```
50 packets transmitted, 50 received, 0% packet loss, time 49071ms
rtt min/avg/max/mdev = 22.215/158.096/319.959/88.832 ms
```

going to check why this is so latent.

> 0.74 seconds does seem very high though so when I get a chance I'll check the behaviour on my test board and see how that compares.

thank you!

trvrnrth, 2021-01-26:
It turns out that delay I was mentioning was actually a whopping 240ms! So on the basis of that I've taken the plunge and ripped out the Bosch Bsec helper class to give me the flexibility to handle that as a timeout callback and properly yield control to the rest of the app and other components. I was avoiding it as it's more hassle but overall the handling is nicer so it's been worth the effort.

The changes are hot off the press over in #31 if you're able to try them out. There's more scope to yield control at certain points during execution if needed but I have a feeling just taking out that massive delay is probably sufficient so they'll no longer be needed. I've only had it running on my test board for a short period of time but initial results are promising.

gretel, 2021-01-27:
@trvrnrth 240ms is quite something! thanks for working on this.

flashed two `esp32` using #31 (19e19cd) and both are working fine. though, it looks - well - similar :)

```
[08:15:39][V][sensor:013]: 'PoniWormBox Temperature': Received new state 18.165531
[08:15:39][VV][sensor.filter:013]: Filter(0x3ffb2bdc)::input(18.165531)
[08:15:39][VV][sensor.filter:053]: MedianFilter(0x3ffb2bdc)::new_value(18.165531)
[08:15:39][V][sensor:013]: 'PoniWormBox Humidity': Received new state 100.000000
[08:15:39][VV][sensor.filter:013]: Filter(0x3ffb3594)::input(100.000000)
[08:15:39][VV][sensor.filter:053]: MedianFilter(0x3ffb3594)::new_value(100.000000)
[08:15:39][V][sensor:013]: 'PoniWormBox Pressure': Received new state 1011.890015
[08:15:39][VV][sensor.filter:013]: Filter(0x3ffb31d0)::input(1011.890015)
[08:15:39][VV][sensor.filter:053]: MedianFilter(0x3ffb31d0)::new_value(1011.890015)
[08:15:39][V][sensor:013]: 'PoniWormBox Gas Resistance': Received new state 98723.000000
[08:15:39][VV][sensor.filter:013]: Filter(0x3ffb3958)::input(98723.000000)
[08:15:39][VV][sensor.filter:053]: MedianFilter(0x3ffb3958)::new_value(98723.000000)
[08:15:39][V][sensor:013]: 'PoniWormBox IAQ': Received new state 190.047607
[08:15:39][VV][sensor.filter:013]: Filter(0x3ffb3d14)::input(190.047607)
[08:15:39][VV][sensor.filter:053]: MedianFilter(0x3ffb3d14)::new_value(190.047607)
[08:15:39][V][sensor:013]: 'PoniWormBox CO2 Equivalent': Received new state 1900.476074
[08:15:39][VV][sensor.filter:013]: Filter(0x3ffb41f8)::input(1900.476074)
[08:15:39][VV][sensor.filter:053]: MedianFilter(0x3ffb41f8)::new_value(1900.476074)
[08:15:39][V][sensor:013]: 'PoniWormBox VOC Equivalent': Received new state 6.060450
[08:15:39][VV][sensor.filter:013]: Filter(0x3ffb4564)::input(6.060450)
[08:15:39][VV][sensor.filter:053]: MedianFilter(0x3ffb4564)::new_value(6.060450)
[08:15:39][V][app:081]: A component took a long time in a loop() cycle (0.73 s).
[08:15:39][V][app:082]: Components should block for at most 20-30ms in loop().
```

did i miss something?

trvrnrth, 2021-01-27:
You should see a few new verbose log lines before that:
```

gretel, 2021-01-27:
oh dang missed to actually copy the updated files over 🤦 

```
[13:33:31][V][bme680_bsec.sensor:170]: Performing sensor run
[13:33:31][VV][i2c:056]: Beginning Transmission to 0x77:
[13:33:31][VV][i2c:093]:     Writing 0b01011010 (0x5A)
[13:33:31][VV][i2c:093]:     Writing 0b01101101 (0x6D)
[13:33:31][VV][i2c:093]:     Writing 0b01100100 (0x64)
[13:33:31][VV][i2c:093]:     Writing 0b01110001 (0x71)
[13:33:31][VV][i2c:061]:     Transmission ended. Status code: 0x00
[13:33:31][VV][i2c:056]: Beginning Transmission to 0x77:
[13:33:31][VV][i2c:093]:     Writing 0b01110100 (0x74)
[13:33:31][VV][i2c:061]:     Transmission ended. Status code: 0x00
[13:33:31][VV][i2c:083]: Requesting 1 bytes from 0x77:
[13:33:31][VV][i2c:113]:     Received 0b01010100 (0x54)
[13:33:31][VV][i2c:056]: Beginning Transmission to 0x77:
[13:33:31][VV][i2c:093]:     Writing 0b01110101 (0x75)
[13:33:31][VV][i2c:061]:     Transmission ended. Status code: 0x00
[13:33:31][VV][i2c:083]: Requesting 1 bytes from 0x77:
[13:33:31][VV][i2c:113]:     Received 0b00000000 (0x00)
[13:33:31][VV][i2c:056]: Beginning Transmission to 0x77:
[13:33:31][VV][i2c:093]:     Writing 0b01110100 (0x74)
[13:33:31][VV][i2c:061]:     Transmission ended. Status code: 0x00
[13:33:31][VV][i2c:083]: Requesting 1 bytes from 0x77:
[13:33:31][VV][i2c:113]:     Received 0b01010100 (0x54)
[13:33:31][VV][i2c:056]: Beginning Transmission to 0x77:
[13:33:31][VV][i2c:093]:     Writing 0b01110010 (0x72)
[13:33:31][VV][i2c:061]:     Transmission ended. Status code: 0x00
[13:33:31][VV][i2c:083]: Requesting 1 bytes from 0x77:
[13:33:31][VV][i2c:113]:     Received 0b00000001 (0x01)
[13:33:31][VV][i2c:056]: Beginning Transmission to 0x77:
[13:33:31][VV][i2c:093]:     Writing 0b01110001 (0x71)
[13:33:31][VV][i2c:061]:     Transmission ended. Status code: 0x00
[13:33:31][VV][i2c:083]: Requesting 1 bytes from 0x77:
[13:33:31][VV][i2c:113]:     Received 0b00010000 (0x10)
[13:33:31][VV][i2c:056]: Beginning Transmission to 0x77:
[13:33:31][VV][i2c:093]:     Writing 0b01110101 (0x75)
[13:33:31][VV][i2c:093]:     Writing 0b00000000 (0x00)
[13:33:31][VV][i2c:093]:     Writing 0b01110100 (0x74)
[13:33:31][VV][i2c:093]:     Writing 0b01010100 (0x54)
[13:33:31][VV][i2c:093]:     Writing 0b01110010 (0x72)
[13:33:31][VV][i2c:093]:     Writing 0b00000001 (0x01)
[13:33:32][VV][i2c:093]:     Writing 0b01110001 (0x71)
[13:33:32][VV][i2c:093]:     Writing 0b00010000 (0x10)
[13:33:32][VV][i2c:061]:     Transmission ended. Status code: 0x00
[13:33:32][VV][i2c:056]: Beginning Transmission to 0x77:
[13:33:32][VV][i2c:093]:     Writing 0b01110100 (0x74)
[13:33:32][VV][i2c:061]:     Transmission ended. Status code: 0x00
[13:33:32][VV][i2c:083]: Requesting 1 bytes from 0x77:
[13:33:32][VV][i2c:113]:     Received 0b01010100 (0x54)
[13:33:32][VV][i2c:056]: Beginning Transmission to 0x77:
[13:33:32][VV][i2c:093]:     Writing 0b01110100 (0x74)
[13:33:32][VV][i2c:093]:     Writing 0b01010101 (0x55)
[13:33:32][VV][i2c:061]:     Transmission ended. Status code: 0x00
[13:33:32][V][bme680_bsec.sensor:201]: Queueing read in 240ms
[13:33:32][VV][scheduler:026]: set_timeout(name='read', timeout=240)
[13:33:32][V][app:081]: A component took a long time in a loop() cycle (0.27 s).
[13:33:32][V][app:082]: Components should block for at most 20-30ms in loop().
[13:33:32][VV][scheduler:152]: Running timeout 'read' with interval=240 last_execution=137131 (now=137371)
[13:33:32][V][bme680_bsec.sensor:206]: Reading data
[13:33:32][VV][i2c:056]: Beginning Transmission to 0x77:
[13:33:32][VV][i2c:093]:     Writing 0b00011101 (0x1D)
[13:33:32][VV][i2c:061]:     Transmission ended. Status code: 0x00
[13:33:32][VV][i2c:083]: Requesting 15 bytes from 0x77:
[13:33:32][VV][i2c:113]:     Received 0b10000000 (0x80)
[13:33:32][VV][i2c:113]:     Received 0b00000000 (0x00)
[13:33:32][VV][i2c:113]:     Received 0b01001100 (0x4C)
[13:33:32][VV][i2c:113]:     Received 0b10100000 (0xA0)
[13:33:32][VV][i2c:113]:     Received 0b11000000 (0xC0)
[13:33:32][VV][i2c:113]:     Received 0b01110100 (0x74)
[13:33:32][VV][i2c:113]:     Received 0b00101100 (0x2C)
[13:33:32][VV][i2c:113]:     Received 0b10000000 (0x80)
[13:33:32][VV][i2c:113]:     Received 0b01100110 (0x66)
[13:33:32][VV][i2c:113]:     Received 0b11101010 (0xEA)
[13:33:32][VV][i2c:113]:     Received 0b10000000 (0x80)
[13:33:32][VV][i2c:113]:     Received 0b00000000 (0x00)
[13:33:32][VV][i2c:113]:     Received 0b00000000 (0x00)
[13:33:32][VV][i2c:113]:     Received 0b00010000 (0x10)
[13:33:32][VV][i2c:113]:     Received 0b10110111 (0xB7)
[13:33:32][V][bme680_bsec.sensor:274]: Publishing sensor states
```

now it's different - takes about 2/3 less time per cycle! 👍

trvrnrth, 2021-01-27:
That's more like the improvement I was expecting!

I suspect you'll find the rest of the time is spent doing the VERY_VERBOSE logging and if you dial it back to VERBOSE you'll be within the acceptable loop block time (that log is itself VERBOSE so it would still be present if applicable).

I'd hope that the combination of these two things will sort out the API disconnects you've been seeing in #29 as well.

gretel, 2021-01-27:
> I'd hope that the combination of these two things will sort out the API disconnects you've been seeing in #29 as well.

now, it's worse 😄 - crashing every cycle even without connecting to the `http` port.

```
[22:42:54][V][api.connection:588]: Hello from client: 'Home Assistant 2021.1.5 (10.0.23.16)'
[22:42:54]Fatal exception 29(StoreProhibitedCause):
[22:42:54]epc1=0x4000df64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
[22:42:54]
[22:42:54]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[22:42:54]
[22:42:54]Exception (29):
WARNING Exception type: Access to invalid address: STORE (wild pointer?)
[22:42:54]epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[22:42:55]
[22:42:55]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[22:42:55]
[22:42:55]ctx: sys
[22:42:55]sp: 3fffec10 end: 3fffffb0 offset: 0190
[22:42:55]3fffeda0:  0000001a 3fff3498 00000002 40100e2c
WARNING Decoded 0x40100e2c: pvPortZalloc
[22:42:55]3fffedb0:  402584f3 00000000 00000005 40258488
WARNING Decoded 0x402584f3: ieee80211_setup_ratetable
WARNING Decoded 0x40258488: ieee80211_setup_ratetable
[22:42:55]3fffedc0:  00000002 4025842f 00000002 40257584
WARNING Decoded 0x4025842f: ieee80211_phy_init
WARNING Decoded 0x40257584: cnx_update_bss_more
[22:42:55]3fffedd0:  402575ad 3fffee80 3fff3498 0000001a
WARNING Decoded 0x402575ad: cnx_update_bss_more
[22:42:55]3fffede0:  40255020 3fffee80 3fff3344 3fff2d30
WARNING Decoded 0x40255020: scan_parse_beacon
[22:42:55]3fffedf0:  3ffef790 3fffee80 3fffee80 40100e08
WARNING Decoded 0x40100e08: free
[22:42:55]3fffee00:  6e616c50 61467465 40006c6c 00000100
[22:42:55]3fffee10:  3ffef0fc 7fffffff 00002200 00000001
[22:42:55]3fffee20:  40251391 00000080 3fffbde4 40101479
WARNING Decoded 0x40251391: ieee80211_parse_beacon
WARNING Decoded 0x40101479: check_poison_neighbors$part$3 at umm_malloc.cpp
[22:42:55]3fffee30:  ffffffbd 3fff2e58 3ffef7a0 3fff3498
[22:42:55]3fffee40:  3fff2300 0000001d 00000000 40255d17
WARNING Decoded 0x40255d17: sta_input
[22:42:55]3fffee50:  00000000 3fff54f4 ffffffbd 00000000
[22:42:55]3fffee60:  00000000 3fff3498 00000013 0000000f
[22:42:55]3fffee70:  40000f58 00000000 08e878f3 00000000
[22:42:55]3fffee80:  00000000 00310909 00640100 00000049
[22:42:55]3fffee90:  3ffef7b4 000000bf 3ffef7cd 3ffef7a8
[22:42:55]3fffeea0:  3ffef7d3 3ffef7b4 3ffef7c0 00000000
[22:42:55]3fffeeb0:  00000000 3ffef841 3ffef7de 3ffef859
[22:42:55]3fffeec0:  3ffef803 3ffef81f 00000000 00000000
[22:42:55]3fffeed0:  00000000 00000000 0000001d 00000000
[22:42:55]3fffeee0:  3fff58cc 4025578a 3fff2300 3fff54f4
WARNING Decoded 0x4025578a: sta_input
[22:42:55]3fffeef0:  00000000 3fff3498 3fff2300 3ffef790
[22:42:55]3fffef00:  3ffef790 000000e7 00000000 0000001d
[22:42:55]3fffef10:  00000000 3ffef79a 4025ef07 3fff2300
WARNING Decoded 0x4025ef07: pp_tx_idle_timeout
[22:42:55]3fffef20:  3ffef784 3fffdcc0 3ffee970 3ffee970
[22:42:55]3fffef30:  00000080 3fff2300 00000000 3ffe85c4
[22:42:55]3fffef40:  4025e7c7 3fffdab0 00000000 40230680
WARNING Decoded 0x4025e7c7: ppPeocessRxPktHdr
WARNING Decoded 0x40230680: loop_task(ETSEventTag*) at core_esp8266_main.cpp
[22:42:55]3fffef50:  3ffee970 40000f49 3fffdab0 40000f49
[22:42:55]3fffef60:  40000e19 00000005 0008347c 00000000
[22:42:55]3fffef70:  3fffefc0 aa55aa55 000000e7 401056fd
WARNING Decoded 0x401056fd: call_user_start_local
[22:42:55]3fffef80:  40105703 0008347c 00000000 2f64322e
WARNING Decoded 0x40105703: call_user_start_local
[22:42:55]3fffef90:  4010000d 322e252f 2d590064 252d6d25
WARNING Decoded 0x4010000d: call_user_start
[... 366 lines cut ...]
[22:43:00]3ffffef0:  00000000 00000000 4bc6a7f0 00000000
[22:43:00]3fffff00:  00000000 4bc6a7f0 40100cbe 90a3d70a
WARNING Decoded 0x40100cbe: millis
[22:43:00]3fffff10:  40105fe1 006867ae 3fff4e6c 00000000
WARNING Decoded 0x40105fe1: ets_timer_arm_new
[22:43:00]3fffff20:  3fff3b80 3fff4e6c 00000010 3fff4e6c
[22:43:00]3fffff30:  0000559e 0000559e 40230700 3fffefa0
WARNING Decoded 0x40230700: esp_yield
[22:43:00]3fffff40:  3fff4e6c 0000559e 0000559e 40231696
WARNING Decoded 0x40231696: __delay
[22:43:00]3fffff50:  3fffb5cc 00000010 3fff433c 4021bac0
WARNING Decoded 0x4021bac0: esphome::Application::loop()
[22:43:00]3fffff60:  00000001 000005e3 3fff6600 3fff65f8
[22:43:00]3fffff70:  3fff65ec 3fffb5cc feefeffe feefeffe
[22:43:00]3fffff80:  00000000 00000000 00000001 3fff4d60
[22:43:00]3fffff90:  3fffdad0 00000000 3fff4d20 4021ee3c
WARNING Decoded 0x4021ee3c: loop
[22:43:00]3fffffa0:  3fffdad0 00000000 3fff4d20 40230824
WARNING Decoded 0x40230824: loop_wrapper() at core_esp8266_main.cpp
[22:43:00]<<<stack<<<
[22:43:00]
[22:43:00]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

still think it's an upstream issue in regards to #29.

gretel, 2021-01-29:
closing, it's an upstream issue. though, thanks a lot for [improving](https://github.com/trvrnrth/esphome-bsec-bme680/pull/31) the loop time!

closed 2021-01-29 by gretel