trvrnrth/esphome-bsec-bme680 #32: Unable to validate config - component not found

eMadman, 2021-01-30:
I see the following error when I try to validate the yaml.

sensor.bme680_bsec: [source /config/esphome/voc_board.yaml:67]
  
Component sensor.bme680_bsec requires component bme680_bsec.

The "\config\esphome\custom_components\bme680_bsec" has been created as described in the readme.  ESPHome and HA have been restarted a few times.  I can't seem to get it to detect the custom component.  Am I missing a step?

Thanks!

trvrnrth, 2021-01-30:
Sounds like you haven't included the component as shown in the configuration examples. The `bme680_bsec:` line in the minimal configuration.

eMadman, 2021-01-30:
That did it.  I had jammed some of those values into the sensors lines

closed 2021-01-30 by eMadman