trvrnrth/esphome-bsec-bme680 #34: Enhancement: Update intervals for temperature and humidity

wifwucite, 2021-02-26:
In ULP mode, the sample rate is 5min, which is fine for the gas sensor that needs heating. Would it be possible to specify a shorter interval like 1min for temperature and humidity?
Btw. I have added a link to this repository in ESPHome‘s BME680 sensor documentation. ;-)

wifwucite, 2021-03-20:
Could you elaborate on that? Where can I set `update_subscription_`? I did not find it.

trvrnrth, 2021-03-20:
This is where the component [sets the sample rate for the temperature sensor](https://github.com/trvrnrth/esphome-bsec-bme680/blob/df37a7635f6acc3f7520f257d2645eda10b863a5/bme680_bsec/bme680_bsec.cpp#L103) for example. At the moment all sensors are either set to `BSEC_SAMPLE_RATE_LP` or `BSEC_SAMPLE_RATE_ULP` depending on your config but it should be possible to use differing update intervals per sensor. In theory you could use any value but I believe the recommendation is to stick to one of those two standard values for the sake of the BSEC algorithm. As I say I've not tried it myself so I'd be interested to know how well it works out.

wifwucite, 2021-04-02:
Thanks, after a git pull (sorry!), I see it, too. ;-) When I configure the global sample rate to ultra low power and I override the sample rate of the pressure or temperature sensor to `BSEC_SAMPLE_RATE_LP`, I get more frequent pressure or temperature readings. Other sample rates, however, seem to confuse it completely, then I do not see any readings at all anymore. But using  `BSEC_SAMPLE_RATE_LP` for the temperature is fine. Of course, we do not know how well heat compensation works with this configuration. Anyhow, thank you very much for your advice.

trvrnrth, 2021-04-02:
Thanks for reporting back! Good to know it works with a ULP/LP combo (and also that other values are a no-go). I'd [read that it should work](https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/Change-the-sample-rate-of-BME680-in-BSEC-library/m-p/12547/highlight/true#M2739) so hopefully the heat compensation is good enough for you.

trvrnrth, 2021-04-26:
@wifwucite I suspect you may have seen already but there's a PR for this feature now at https://github.com/esphome/esphome/pull/1710. This also improves the heat compensation as it turns out the timestamp being supplied for the inputs should be the trigger time not the measurement time (a bug that appears to exist in some of the bosch examples too).

wifwucite, 2021-04-26:
Thanks for the update, yes, I have seen the PR. I was not sure if it will impact this custom component as well. Or will this PR be merged into ESPHome?

trvrnrth, 2021-04-26:
This component will be in the next core release of ESPHome so in future won't need adding as a custom component. Although I may keep this repo around anyway as it does no harm and allows for faster iterations if necessary. In the meantime as the next release date of ESPHome is unknown I'll probably formalise and release those changes here too.

wifwucite, 2021-04-26:
Sounds good! And thank you for the clarification.

trvrnrth, 2021-05-04:
Now released as [1.2.0](https://github.com/trvrnrth/esphome-bsec-bme680/releases/tag/v1.2.0).

closed 2021-05-04 by trvrnrth