trvrnrth/esphome-bsec-bme680 #35: Implausible Measurements

lukasdietz, 2021-02-28:
Hey there,

I am trying to get a bme680 sensor up and running with this implementation of the bsec software. 
I am able to compile the code just fine but it seems to be impossible to get accurate values. 
After opening my Window for around 30 minutes values for co2 equivalent are almost at 2500 ppm and for VOC equivalent at around 10 ppm. This seems to be implausible to me because of the opened window.

Can someone tell me how to properly calibrate this sensor to get accurate readings from it? 
Is there an option to force a recalibration? 

I am hoping somebody can help me out.

caddyblack, 2021-02-28:
Hi Lucas,

I have a MH-Z19 IR CO2 measurement near the bme680 to compare the results.
The resistivity of the bme680 was very low in the beginning. First week not
even s ove 30000 Ohm. The calculated CO2 was always far off.
After about 2 months the bme680 has stabilised. In clean air I get
resistivity values of 300000 Ohm with a calculated CO2 of 400 ppm.
I have tried with a high temperarure setting (400°C) but that did not help.
Now I have 1043 ppm CO2 on the bme680 and 743 ppm on the MH-Z19.
The VOC shows 9.8 ppm, a plausible value.
So use the bme680 for rough estimates only. The voc seems KK.
air pressure, humidity and temperature are very stable on the bme680 but
the offset of the temperature is 3.8°C!
Once tested and calibrated I only use it for the pressure, humidity and
temperature.
For CO2 measurements the bme680 is not accurate enough. You can use is to
determine rough estimates like 400-1500 ppm for normal indoor air and >2500
ppm you need extra airflow/ventilation.

Hope this information helpx you to judge the values you see.

lukasdietz, 2021-02-28:
Hi,
I am glad someone did help me so fast! 
Thanks for your information. I am trying to build a "Covid ventilation alarm". Acoustic signal whenever it is needed to open a window. 
So if I understand you right I will have to give the sensor a quite long time to stabilize and use it for fairly accurate readings?
Would you say the IAQ values are accurate enough to make such an "alarm system"? 
Currently I am sitting around 200 for fresh air... 

[image]

These are my measurements for the last few hours.

caddyblack, 2021-02-28:
Hi Lukas,

Please find my recent history graphs for comparison.

lukasdietz, 2021-03-01:
Sorry for this question. Where do I find your history graphs?

caddyblack, 2021-03-01:
> Sorry for this question. Where do I find your history graphs? 

See my previous answer with the links

caddyblack, 2021-03-01:
> Sorry for this question. Where do I find your history graphs?

Reply in Github. Are pictures visible now?
[image]
[image]
[image]
[image]
[image]
[image]
Green is the MH-Z19 data, Yellow the BME680 calculated CO2 data.

lukasdietz, 2021-03-02:
Yes now we can see your graphs!
Thanks for your help. Since yesterday the values seem to stabilize in a realistic value region.
Now I get IAQ values around 30.

trvrnrth, 2021-03-09:
Interesting to see the comparison graphs. Doesn't seem like too bad an approximation once stable!

closed 2021-03-09 by trvrnrth

rubinho101, 2022-01-14:
> Hi Lucas, I have a MH-Z19 IR CO2 measurement near the bme680 to compare the results. The resistivity of the bme680 was very low in the beginning. First week not even over 30000 Ohm. The calculated CO2 was always far off. After about 2 months the bme680 has stabilised. In clean air I get resistivity values of 300000 Ohm with a calculated CO2 of 400 ppm. I have tried with a high temperarure setting (400°C) in the first weeks but that did not help. Now I have 1043 ppm CO2 on the bme680 and 743 ppm on the MH-Z19. The VOC shows 9.8 ppm, a plausible value. So use the bme680 for rough estimates only. The voc seems OK. air pressure, humidity and temperature are very stable on the bme680 but the offset of the temperature is 3.8°C on my device! Once tested and calibrated I only use it for the pressure, humidity and temperature. For CO2 measurements the bme680 is not accurate enough. You can use is to determine rough estimates like 400-1500 ppm for normal indoor air and >2500 ppm you need extra airflow/ventilation. Hope this information helps you to judge the values you see. Op zo 28 feb. 2021 18:15 schreef lukasdietz <notifications@github.com>:
> […](#)

Would you recommend BME680 for VOC? What do you suggest for VOC?