trvrnrth/esphome-bsec-bme680 #36: Boot Loop when flashing the custom component

DonRichie, 2021-03-25:
Hello, 

Unfortunately I can't make the custom component work. 
After flashing the new firmware the device enters a boot loop until it is falling back to safe mode.

**This is working without any problems:**
```
i2c:
 - id: bus_a
   sda: GPIO14
   scl: GPIO12
   scan: False
   
sensor:
  - platform: bme680
    i2c_id: bus_a
    address: 0x77
    update_interval: ${bme680_update_interval}

  # BME680 RAW DATA
    temperature:
      name: "${device_name_frontend} Temperature"
      id: "bme680_${device_name}_temperature"
    pressure:
      name: "${device_name_frontend} Pressure"
      id: "bme680_${device_name}_pressure"
    humidity:
      name: "${device_name_frontend} Relative Humidity"
      id: "bme680_${device_name}_relative_humidity"
    gas_resistance:
      name: "${device_name_frontend} Gas Resistance" 
      id: "bme680_${device_name}_gas_resistance"
```

**This doesn't work. The git repo was cloned to the directory bme680_bsec right next to the yaml file:**
```
i2c:
 - id: bus_a
   sda: GPIO14
   scl: GPIO12
   scan: False

bme680_bsec:
    i2c_id: bus_a
    address: 0x77

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
    humidity:
      name: "BME680 Humidity"
```

Does someone have an indea how to debug this?

The Bootloop gives an error via serial console which might be decodable with platform.io:
```
[08:25:27]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[08:25:27]
[08:25:27] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[08:25:27]
[08:25:27]load 0x4010f000, len 3584, room 16 
[08:25:27]tail 0
[08:25:27]chksum 0xb0
[08:25:27]csum 0xb0
[08:25:27]v2843a5ac
[08:25:27]~ld
[08:25:27][I][logger:166]: Log initialized
[08:25:27][C][ota:366]: There have been 9 suspected unsuccessful boot attempts.
[... 297 lines cut ...]
WARNING Decoded 0x40231b70: precache
WARNING Decoded 0x40231b70: precache
[08:25:28]3fffff0c:  40231b70 00000000 00000000 3fff3698  
WARNING Decoded 0x40231b70: precache
[08:25:28]3fffff1c:  00000000 00000000 3fff36b0 00000000  
[08:25:28]3fffff2c:  00000000 00000000 40231b70 00000000  
WARNING Decoded 0x40231b70: precache
[08:25:28]3fffff3c:  40231b70 00000000 40231b70 00000000  
WARNING Decoded 0x40231b70: precache
WARNING Decoded 0x40231b70: precache
[08:25:28]3fffff4c:  40231b70 00000000 40231b70 00000000  
WARNING Decoded 0x40231b70: precache
WARNING Decoded 0x40231b70: precache
[08:25:28]3fffff5c:  00000000 3fff3cec 3fff3f8c 3fff4024  
[08:25:28]3fffff6c:  3fff404c 3fff3e94 3fff3d4c 3fff3bf4  
[08:25:28]3fffff7c:  feefeffe 3fff404c 00000060 3fff40ac  
[08:25:28]3fffff8c:  feefeffe feefeffe feefeffe feefeffe  
[08:25:28]3fffff9c:  3fff2250 3fffdad0 00000000 3fff2210  
[08:25:28]3fffffac:  4022c8a0 feefeffe feefeffe 3ffe8574  
WARNING Decoded 0x4022c8a0: loop_wrapper() at core_esp8266_main.cpp
[08:25:28]<<<stack<<<
[08:25:28]
[08:25:28]--------------- CUT HERE FOR EXCEPTION DECODER ---------------

```

trvrnrth, 2021-03-29:
Do you have only the BME680 component and sensors in your config or do you have others set up as well that could be contributing to memory/stack usage?

DonRichie, 2021-04-01:
Damn you are right. The memory was exhausted.
I had a small WS2812B strip configured, too. After removing it everything works as expected.

Thank you for your help.

closed 2021-04-01 by DonRichie

DonRichie, 2021-04-02:
@trvrnrth 
I don't think this issue was a memory issue.
I changed to an ESP32 and the bootloop still happened.

Using Neopixel and bsec_bme680 at the same time does not work for me. Only one of them works.
I am aware about your answer in issue #37 . I believe I suffer from the same bootloop. I am glad developers are on it.

Will use two ESP8266 as a workaround now. This gives me additional 0.6 W permanent Energy usage, but it will be the least pain.