trvrnrth/esphome-bsec-bme680 #37: Not able to use bsec and adc platform simultaneously

zenzay, 2021-03-26:
Hi.

First of all; thanks for the software!

I had plans of attaching a MQ-2 gas sensor - besides the BME680 - to a TTGO ESP32 board, but I'm getting boot loops as soon as I used both the bme680_bsec and adc platforms. Both works fine by themselves, but not together. I have tried different adc pins on the board, to no avail. Any pointers to why this happens and a possible work-around?

```
[21:31:11][I][logger:166]: Log initialized
[21:31:11][C][ota:366]: There have been 9 suspected unsuccessful boot attempts.
[21:31:11][I][app:029]: Running through setup()...
[21:31:11][C][spi:022]: Setting up SPI bus...
[21:31:11][C][bme680_bsec.sensor:018]: Setting up BME680 via BSEC...
[21:31:11]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[21:31:11]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[21:31:11]Core 1 register dump:
[21:31:11]PC      : 0x4013bf03  PS      : 0x00060936  A0      : 0x80134d59  A1      : 0x3ffb0070  
[21:31:11]A2      : 0x3ffb80c0  A3      : 0x3ffb03b0  A4      : 0x3ffb0294  A5      : 0x00000010  
[21:31:11]A6      : 0x00000003  A7      : 0x3ffb02d4  A8      : 0xa0000000  A9      : 0x00000000  
[21:31:11]A10     : 0x00000000  A11     : 0x10000000  A12     : 0x00000000  A13     : 0x3ffb0b00  
[21:31:11]A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
[21:31:11]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[21:31:11]
[21:31:11]Backtrace: 0x4013bf03:0x3ffb0070 0x40134d56:0x3ffb00a0 0x40131841:0x3ffb03b0 0x400e2bdd:0x3ffb0470 0x400d8bf9:0x3ffb04e0 0x400d8dc7:0x3ffb0db0 0x400d8e0a:0x3ffb0fa0 0x40172b6d:0x3ffb0fc0 0x40172c29:0x3ffb0fe0 0x400e208d:0x3ffb1000 0x400e8036:0x3ffb1050 0x400f99e3:0x3ffb1fb0 0x40089cc9:0x3ffb1fd0
[21:31:11]
[21:31:11]Rebooting...
```

trvrnrth, 2021-03-29:
This looks like it might be fixed by increasing the loop stack size. Unfortunately that's not entirely straightforward at the moment but there's an ongoing discussion in https://github.com/esphome/issues/issues/1875

zenzay, 2021-03-31:
Thanks! I'll check that discussion out. Once again; thanks for the amazing software.

closed 2021-03-31 by zenzay