trvrnrth/esphome-bsec-bme680 #38: Component not found: v    gas_resistance.

skywluke, 2021-04-07:
Hi! First of all thank you for your effort! I had a problem during validate: Failed config v    gas_resistance:   Component not found: v    gas_resistance. What is your oppinion? Thanks!

trvrnrth, 2021-04-07:
If you post your config I can probably point out the issue.

skywluke, 2021-04-07:
The BME680 one and only device on wemos d1 mini pro. I just copy everything from the your "Advanced configuration". Before that, these lines in the config:

esphome:
  name: wemos_3
  platform: ESP8266
  board: d1_mini

wifi:
manual_ip:
captive_portal:
logger:

i2c:
  scl: D1
  sda: D2
  scan: True
  id: bus_a

Many thanks!

trvrnrth, 2021-04-07:
That should be fine then. Could you triple check the indenting?

skywluke, 2021-04-08:
It was my mistake, Holded the previous config until the new isn't working with hash mark. And one of hashmark was "v" instead of #. Sorry.... and many thanks for support!

trvrnrth, 2021-04-08:
Glad you got it sorted :)

closed 2021-04-08 by trvrnrth

skywluke, 2021-04-11:
Yes, when i cheked the indenting i found the missed config piece. Many Thanks!