trvrnrth/esphome-bsec-bme680 #39: Possible to reset state?

JohnnyM84, 2021-04-09:
Is it possible to reset the saved state of the bme680, i have moved it around when i was building it and now the iaq and eco2, stay having high values (120+ iaq and 1100+ eco2). What i read this must be stablizing in a day but that isn't happening so i want to start over bij ressetting the state.
Is this possible?

trvrnrth, 2021-04-09:
The state is stored in flash via the ESPHome preferences system. I can't recall off the top of my head if they are cleared with an OTA update but a full serial re-flash should certainly clear it out for you.

I would expect it to sort itself out eventually anyway though as whilst the saved state is used as a starting point, if the environment is different then re-calibration should occur regardless. The time taken will depend on your environment and it's stability so I don't think any particular guarantuees are made about how long it will take.

JohnnyM84, 2021-04-09:
I already did an ota where i removed the bme680 config on de esp, and then i enabled it again but it started again high where i espected to start low and then relearn it.

But is it in the memory of the esp board saved?

trvrnrth, 2022-08-10:
As a serial flash caters for this scenario I'm going to close out this issue. I would almost be of the opinion that a core ESPHome feature to clear all saved preferences might be the best solution if a new feature request were to be raised.

closed 2022-08-10 by trvrnrth