trvrnrth/esphome-bsec-bme680 #40: bsec-bme680 not working in esphome 0.17.0

r100gs, 2021-05-04:
Hello,

when I try to compile my code in new esphome 0.17.0 I get following error:

`INFO Reading configuration /config/esphome/dachschlazitemp.yaml...
ERROR Unable to load custom component bme680_bsec.sensor:
Traceback (most recent call last):
  File "/opt/esphome/esphome/config.py", line 141, in _lookup_module
    module = importlib.import_module(f"custom_components.{domain}")
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/config/esphome/custom_components/bme680_bsec/sensor.py", line 34, in <module>
    sensor.sensor_schema(UNIT_CELSIUS, ICON_THERMOMETER, 1),
TypeError: sensor_schema() missing 1 required positional argument: 'device_class_'
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 27 March 02:00:00 to 30 October 03:00:00
Failed config

sensor.bme680_bsec: [source /config/esphome/dachschlazitemp.yaml:82]
  
  Platform not found: 'sensor.bme680_bsec'.
`

I checked the directory. It is in right place /config/esphome/custom_components/

With previous version 0.16.3 it was working fine. Now it cannot be compiled.

Best regards,
Stefan

trvrnrth, 2021-05-04:
Ah, that went out with that breaking change then...

[Release 1.2.0](https://github.com/trvrnrth/esphome-bsec-bme680/releases/tag/v1.2.0) should resolve that for you.

r100gs, 2021-05-04:
That was fast! Works! THX

closed 2021-05-04 by r100gs