trvrnrth/esphome-bsec-bme680 #42: Esphome 1.18.0

JohnnyM84, 2021-05-19:
In esphome 1.18.0 is Now an bme680 bse available, is this the same component as this repository?

i see a difference in sample rate configuration.

trvrnrth, 2021-05-19:
Yup. ESPHome 1.18.0 includes release 1.1.0 with 1.20 slated for inclusion probably in ESPHome 1.19. In the meantime you can keep using the component from this repo to take advantage of the extra sample rate config. I will at some point update the readme to make this clearer.

JohnnyM84, 2021-05-19:
Oké, thanks for the information

JohnnyM84, 2021-05-20:
I tried the new code and set the sample rate to ulp and for temp, humidity and pressure on lp but after 20 mins i have still no gas readings.

trvrnrth, 2021-05-20:
Could you paste your config here?

JohnnyM84, 2021-05-20:
i got the readings now 1 time in 50 mins.
This is my config:
`web_server:
  port: 80
  
i2c:
  sda: 2
  scl: 0
  
bme680_bsec:
    # i2c address
    # -----------
    # Common values are:
    # - 0x76
    # - 0x77
    # Default: 0x76
    address: 0x77

    # Temperature offset
    # ------------------
    # Useful if device is in enclosure and reads too high
    # Default: 0
    temperature_offset: 2

    # IAQ calculation mode
    # --------------------
    # Available options:
    # - static (for fixed position devices)
    # - mobile (for on person or other moveable devices)
    # Default: static
    iaq_mode: static

    # Sample rate
    # -----------
    # Available options:
    # - lp (low power - samples every 3 seconds)
    # - ulp (ultra low power - samples every 5 minutes)
    # Default: lp
    sample_rate: ulp

    # Interval at which to save BSEC state
    # ------------------------------------
    # Default: 6h
    state_save_interval: 6h

sensor:
  - platform: bme680_bsec
    temperature:
      # Temperature in °C
      name: "Slaapkamer BME680 Temperature"
      sample_rate: lp
      filters:
        - median
    pressure:
      # Pressure in hPa
      name: "Slaapkamer BME680 Pressure"
      sample_rate: lp
      filters:
        - median
    humidity:
      # Relative humidity %
      name: "Slaapkamer BME680 Humidity"
      sample_rate: lp
      filters:
        - median
    gas_resistance:
      # Gas resistance in Ω
      name: "Slaapkamer BME680 Gas Resistance"
      filters:
        - median
    iaq:
      # Indoor air quality value
      name: "Slaapkamer BME680 IAQ"
      filters:
        - median
    iaq_accuracy:
      # IAQ accuracy as a numeric value of 0, 1, 2, 3
      name: "Slaapkamer BME680 Numeric IAQ Accuracy"
    co2_equivalent:
      # CO2 equivalent estimate in ppm
      name: "Slaapkamer BME680 CO2 Equivalent"
      filters:
        - median
    breath_voc_equivalent:
      # Volatile organic compounds equivalent estimate in ppm
      name: "Slaapkamer BME680 Breath VOC Equivalent"
      filters:
        - median

  - platform: uptime
    name: Uptime Seconds
    id: uptime_sensor
    update_interval: 60s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? String(days) + "d " : "") +
                (hours ? String(hours) + "h " : "") +
                (minutes ? String(minutes) + "m " : "") +
                (String(seconds) + "s")
              ).c_str();

  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      # IAQ accuracy as a text value of Stabilizing, Uncertain, Calibrating, Calibrated
      name: "Slaapkamer BME680 IAQ Accuracy"
  - platform: template
    name: Uptime Human Readable
    id: uptime_human
    icon: mdi:clock-start`

JohnnyM84, 2021-05-20:
can't get the formating good

trvrnrth, 2021-05-20:
I've formatted your comment. The reason for the delay is the [median filter](https://esphome.io/components/sensor/index.html#median). You can adjust or remove it to your liking depending on your requirements.

JohnnyM84, 2021-05-20:
Yes that was the problem actually logical, witk 5 min interval median is not needed.
for temp, humidity and pressure i have still median.

the accauracy for temp and humidity in this mode is a litte bit better.

closed 2021-05-20 by trvrnrth