trvrnrth/esphome-bsec-bme680 #43: Communication failed

loefchen, 2021-05-19:
I'm fiddeling around with the configuration, but I'm permanently getting an error:

[bme680_bsec.sensor:135]: Communication failed (BSEC Status: 0, BME680 Status: -2)

It seems like one of my tries produced some values, but as I recognized it, it was too late and I could not reproduce it anymore.

I'm using a Wemos D1 mini with following I2C-config:

```
i2c:
  sda: 4
  scl: 5
```

The following config is is the example config from the readme.

Could anyone help?

loefchen, 2021-05-19:
got it... Changed the adress from 0x76 to 0x77...

weird, I thought 0x76 should be standard.

closed 2021-05-19 by loefchen

trvrnrth, 2021-05-20:
It'll depend on your board. Some appear to default to 0x76, some to 0x77.

Anyway, glad you got it sorted!