trvrnrth/esphome-bsec-bme680 #49: bme680 device does not connect to wifi anymore

r100gs, 2021-11-09:
``Hello,

I have following config, which was working fine with custom components.
But now I compiled it again with homeassistant 2.11.1. It compiles fine and I can load it to my device.
The problem is, that it does not connect to wifi anymore.
Maybe s. b. can have a look at my bme680 config if its ok.

`esphome:
  name: dachschlazitemp
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "xxx"
  password: "xxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxxx"
    password: "xxxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxx"
  encryption:
    key: xxxx

ota:
  password: "xxx"

i2c:
  - id: bus_a
    sda: 04 #D2
    scl: 05 #D1
    scan: True

bme680_bsec:
    # i2c address override (default is 0x76)
    address: 0x77

    # Temperature offset if device is in enclosure and reads too high (default is 0)
    temperature_offset: 0.0

    # Mode for IAQ sensors if device is mobile (default is static)
    iaq_mode: static

    # Interval at which to save BSEC state (default is 6 hours)
    state_save_interval: 6h

    # Sample rate
    # -----------
    # Available options:
    # - lp (low power - samples every 3 seconds)
    # - ulp (ultra low power - samples every 5 minutes)
    # Default: lp
    sample_rate: ulp

time:
  - platform: homeassistant
    id: homeassistant_time

sensor:
  - platform: uptime
    name: Dach SchlaZi Uptime sec
    id: uptime_sec
    update_interval: 60s
  - platform: wifi_signal
    name: "WiFi Signal Dach SchlaZi"
    update_interval: 60s

## bme680_bsec
  - platform: bme680_bsec
    temperature:
      name: "Dach SchlaZi Temperatur"
      id: temp
      sample_rate: lp
      filters:
        - median

    pressure:
      name: "Dach SchlaZi Luftdruck"
      id: druck
      sample_rate: lp
      unit_of_measurement: "hPa"
      filters:
        - lambda: >-
            return x / powf(1.0 - (415 / 44330.0), 5.255);

    humidity:
      name: "Dach Schlazi Luftfeuchtigkeit"
      id: feuchte
      sample_rate: lp
      filters:
        - median

    gas_resistance:
      name: "Dach SchlaZi CO2 Widerstand"
      id: gas
      filters:
        - median
    iaq:
      name: "Dach SchlaZi BME680 IAQ"
      id: iaq_range
      filters:
        - median

    co2_equivalent:
      name: "Dach SchlaZi BME680 CO2 Equivalent"
      filters:
        - median
  
    breath_voc_equivalent:
      name: "Dach SchlaZiBME680 Breath VOC Equivalent"
      filters:
        - median

    iaq_accuracy:
      # IAQ accuracy as a numeric value of 0, 1, 2, 3
      name: "Dach SchlaZi BME680 Numeric IAQ Accuracy"

## Berechnungen 
 
 
  - platform: template
    name: "Dach SchlaZi Absolute Feuchtigkeit"
    lambda: |-
      const float mw = 18.01534;    // molar mass of water g/mol
      const float r = 8.31447215;   // Universal gas constant J/mol/K
      return (6.112 * powf(2.718281828, (17.67 * id(temp).state) /
        (id(temp).state + 243.5)) * id(feuchte).state * mw) /
        ((273.15 + id(temp).state) * r); // in grams/m^3
    accuracy_decimals: 2
    update_interval: 60s
    icon: 'mdi:water'
    unit_of_measurement: 'g/m³'
    
  - platform: template
    name: "Dach SchlaZi Luftdruck auf Meereshöhe"
    lambda: |-
      const float STANDARD_ALTITUDE = 415; // in meters, see note
      return id(druck).state / powf(1 - ((0.0065 * STANDARD_ALTITUDE) /
        (id(temp).state + (0.0065 * STANDARD_ALTITUDE) + 273.15)), 5.257); // in hPa
    update_interval: 60s
    unit_of_measurement: 'hPa'

  - platform: template
    name: "Dach SchlaZi Taupunkt"
    lambda: return (243.5*(log(id(feuchte).state/100)+((17.67*id(temp).state)/
     (243.5+id(temp).state)))/(17.67-log(id(feuchte).state/100)-
     ((17.67*id(temp).state)/(243.5+id(temp).state))));
    update_interval: 60s
    unit_of_measurement: '°C'
    
binary_sensor:
- platform: status
  name: "Dach SchlaZi Temp Status"
    
text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
    ssid:
      name: ESP Connected SSID
    bssid:
      name: ESP Connected BSSID
    mac_address:
      name: ESP Mac Wifi Address

# bme680_sec      
  - platform: bme680_bsec
    iaq_accuracy:
      name: "Dach SchlaZi BME680 IAQ Accuracy"
      
      
  - platform: template
    name: Dach SchlaZi Uptime
    lambda: |-
      int seconds = (id(uptime_sec).state);
      int days = seconds / (24 * 3600);
      seconds = seconds % (24 * 3600); 
      int hours = seconds / 3600;
      seconds = seconds % 3600;
      int minutes = seconds /  60;
      seconds = seconds % 60;
      return { (String(days) +"d " + String(hours) +"h " + String(minutes) +"m "+ String(seconds) +"s").c_str() };
    icon: mdi:clock-start
    update_interval: 65s
    
    
    
  - platform: template
    name: "Dach SchlaZi Luftqualität"
    icon: "mdi:air-filter"
    lambda: |-
   
      ESP_LOGD("main", "Aktueller Wert %d", id(iaq_range));
      
      if ((id(iaq_range).state <= 50.0 )) {
        return {"sehr gut"};
      }
      else if ((id(iaq_range).state > 50.00)  && (id(iaq_range).state <= 100.0)) {
        return {"gut"};
      }
      else if ((id(iaq_range).state > 100.0) && (id(iaq_range).state <= 150.0)) {
        return {"Leicht verschmutz"};
      }
      else if ((id(iaq_range).state > 150.0) && (id(iaq_range).state <= 200.0)) {
        return {"mäßig verschmutz"};
      }
      else if ((id(iaq_range).state > 200.0) && (id(iaq_range).state <= 250.0)) {
        return {"stark verschmutzt"};
      }
      else if ((id(iaq_range).state > 250.0) && (id(iaq_range).state <= 350.0)) {
        return {"extrem verschmutz"};
      }
      else if ((id(iaq_range).state > 350.00)) {
        return {"gefährlich verschmutzt"};
      }
      return {};
    update_interval: 60s
    
`

caddyblack, 2021-11-09:
Hi,

Is your API encryption key a 32 bit base64 encoded string?
Have you tried without the encryption?

Caddyblack

r100gs, 2021-11-09:
Hello caddyblack,

thx for the suggestion.
I use the key with my other devices and everything is ok.
It is only the device with the bme680 which does not work ( I have only one bme680).
BME280 devices works good (same key)

Best regards,
Stefan

trvrnrth, 2022-08-10:
If this is still happening for you on the latest version, could you post a new issue at https://github.com/esphome/issues/issues and include the full serial logs showing the wifi connection attempts etc.

closed 2022-08-10 by trvrnrth