trvrnrth/esphome-bsec-bme680 #50: bme688 fails

michaelm-github, 2021-12-13:
With a bme688, gas resistance stays at 12946860, the initialization value.

gretel, 2022-01-24:
same issue here. 
[image]
@trvrnrth be cool if we could work this out!
maybe also work on https://github.com/esphome/feature-requests/issues/1332? regards

awstanley, 2022-06-10:
Not sure if this is of any help/use to anyone, but in programming it away from the ESPHome environment I run self-test from the standard API as part of the initalisation.  The self-test soft resets the system and the gas resistance works; without the self-test the gas resistance can get stuck on both the common Arduino library and using the BME68x driver BoschSensorTec's github provides.

The bigger issue to consider with the BME688 is t_fine needs to be set in most instances as even sensors from the same batch are different (beyond what the specifications sheet might indicate).  This becomes more important if you're looking to support custom gas resistance values (without using BSEC) as self-heating in different enclosures can create issues.  It's unlikely that this is the underlying cause of the gas resistance freezing, and it's more likely something weird going on with soft resets or calibration data needing to be reset under certain circumstances.

trvrnrth, 2022-08-10:
I'm going to close this out in favour of the discussion/work over at https://github.com/esphome/feature-requests/issues/1332. I'm afraid I have neither the devices (nor really the time) to provide any useful input myself at this point.

closed 2022-08-10 by trvrnrth